aircrack-ng/rtl8188eus #299: TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS] | Unable to find in iwconfig

proboman007, 2024-10-30:
Hi, 
When I first plugin my rtl8188eus usb it was visible in iwconfig command after update && upgrade the kali linux, but when I download the drivers from https://github.com/aircrack-ng/rtl8188eus.git and try to run make command it gave the following error.

Now it is not showing with iwconfig command but when I am trying lsusb command it shows that usb is pluged-in.

Following is the error:

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.11.2-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: *** /lib/modules/6.11.2-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2065: modules] Error 2

gglluukk, 2024-10-30:
```
apt --yes install build-essential linux-headers-`uname -r`
```

bigpiss-mp3, 2024-11-01:
> ```
> apt --yes install build-essential linux-headers-`uname -r`
> ```

my dude thank you so much just spent the last 10 hours trying to figure this out

gglluukk, 2024-11-01:
your pain is mine

otiescereal, 2024-12-27:
hey man i"m new to the kali world and just got one of those adapters too TL-WN722N v4 any chance you can explain how to make it work properly but in like baby steps still gettin used to these comands