aircrack-ng/rtl8188eus #304: Plugging in RTL8811EUS results in system hard freeze after `bFWReady == _FALSE call reset 8051...` and `==> rtl8188e_iol_efuse_patch`.

dreirund, 2024-11-14:
Ahoj,

installing this (building branch 5.3.9, using [this recipe](https://aur.archlinux.org/packages/8188eu-aircrack-dkms-git)) on Linux 6.11.7, and plugging in a RTL8811EUS device, results in hard freeze (even `SysRq` not working anymore).

Last messages I see in a running `dmesg -w` before system freezes:

```
[...]
bFWReady == _FALSE call reset 8051...
usbcore: registered new interface driver 8188eu
==> rtl8188e_iol_efuse_patch
```  
*(and then system hard freezed.)*

Screen photograph:

---

[image]

---

My kernel config: [`kernel-config.txt`](https://github.com/user-attachments/files/17756885/kernel-config.txt)

Regards!

gglluukk, 2024-11-29:
@dreirund @Blu3cr0ss 
please try this branch: 
https://github.com/gglluukk/rtl8188eus

@dreirund could you try gcc for kernel's compilation too?

dreirund, 2024-11-29:
> @dreirund @Blu3cr0ss please try this branch: https://github.com/gglluukk/rtl8188eus

I decided to not mess around with this device anymore.

> @dreirund could you try gcc for kernel's compilation too?

Not anymore; I decided to not mess around with this device anymore due to the difficulties of getting it to work.

gglluukk, 2024-11-29:
@dreirund 
btw you can use `rtl8xxxu` already in kernel with no headache

dreirund, 2024-11-29:
> btw you can use `rtl8xxxu` already in kernel with no headache

Did not work at all for me with my device.