aircrack-ng/rtl8188eus #307: warning: the compiler differs from the one used to build the kernel

diehard88, 2024-12-04:
I recently upgraded from Linux Mint 21.3 to 22.   The kernel was also upgraded to Linux 6.8.

Now I´m getting this warning if I compile the rtl8188eus driver.

`~/rtl8188eus$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.0-47-generic/build M=/home/xpto/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-47-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-47-generic'`

any ideas?

gglluukk, 2024-12-04:
it's ok, compilers' versions are same, only calling binaries are different

diehard88, 2024-12-05:
> it's ok, compilers' versions are same, only calling binaries are different

understood.

before the upgrade, compilation would take several seconds and display a lot of messages during ´make´ and ´sudo make install´ and end successfully.

now it simply shows that warning, and nothing else.   

the ´sudo make install´ shows the following:    

[image]

diehard88, 2024-12-15:
I just realized that there is no problem.

I was using @gglluukk rtl8188eu fork to compile the drivers.
his version displays more info during compilation, while the original 8188eu is more succinct.

closed 2024-12-15 by diehard88

gglluukk, 2024-12-15:
difference is that https://github.com/gglluukk/rtl8188eus supports 6.8 but pull requests have never been accepted