aircrack-ng/rtl8188eus #308: linux kernel 6.8.0.49 -- 8188eu: disagrees about version of symbol skb_pull

diehard88, 2024-12-07:
the 8188 wireless interface doesn´t exist, though the realtek 8188 adapter is detected.

[image]

this is what happens during bootup.

[image]

this doesn´t happen with kernel 6.8.0.47.      

have you seen this before?

diehard88, 2024-12-07:
[image]

gglluukk, 2024-12-07:
try this:
https://github.com/gglluukk/rtl8188eus

diehard88, 2025-03-17:
> try this: https://github.com/gglluukk/rtl8188eus

I've been using your fork with great success, thanks.

I just noticed that the drivers of this repo (from which you forked), are deprecated.

moving forward, one should use github.com/lwfinger/rtw88.....

will you create a new fork?